whoenig/crazyflie_ros #182: Unable to get any Orientation or Covariance data in IMU topic

kashishdhal, 2020-06-01:
Hello @whoenig ,

I am not getting any orientation or covariance data in IMU topic. Please help!

Regards,
Kashish Dhal

whoenig, 2020-06-01:
The IMU does not measure orientation, but angular velocity and acceleration. For both, the exact noise model is not part of crazyflie_ros and the covariance is simply set to -1 as per ROS standard, see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp#L608-L631. You can find noise estimates in state estimators (e.g., https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/kalman_core.c#L142-L149) and research papers (e.g., the Crazyswarm paper had some data on the gyro noise for CF 2.0).

Perhaps you are instead (or in addition) interested in the output of the state estimator (e.g., the EKF)? In that case, please take a look at the `/pose` topic that contains the actual orientation estimate based on sensor fusion. The covariance is estimated in the firmware, but by default not exposed in crazyflie_ros. You can create a custom log, that captures the relevant log variables defined in https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/estimator_kalman.c#L720-L728.

kashishdhal, 2020-06-01:
Hello @whoenig ,

Yes, I am interested in EKF outputs. Actually, in the IMU topic, there are fields for orientation and covariance, etc. but they are blank. I will take a look on `\pose` topic as per your suggestion.

So if I create a custom log, then how will it be transmitted through radio? Do I need to do some modifications in both ros stack and crazyflie firmware? Also, if I want to send the motor commands directly through the ros stack then how can we do that? I realize that on-board controller will have to be in the loop for stability.

Can you please point me to the piece of code where we are sending the radio commands from ros stack and where these commands are recieved in the crazyflie firmware? It will be a great help to me. I just want to understand the flow of the code.

Regards,
Kashish Dhal

whoenig, 2020-06-02:
- IMU topic: We rely on the standard IMU topic from ROS. However, for the crazyflie we do not have data for orientation or covariance.
- There are no changes needed in the ROS stack or firmware, but you need to configure ROS accordingly. You can find an example of custom log blocks here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/customLogBlocks.launch#L14-L19. The names of the variables have to be obtained from the firmware (there is no compiled list, yet), but the ones you are looking for are the ones I linked to, e.g., 'kalman.varD0'. Note that the log system is optimized for low latency, so you cannot request all information that you might need simultaneously. You can find some more information at https://www.bitcraze.io/documentation/repository/crazyflie-firmware/2019.09/logparam/.
- On the ROS side the most low-level part that sends data is https://github.com/whoenig/crazyflie_cpp/blob/master/src/Crazyradio.cpp#L136-L267. There are multiple layers of abstraction on top of that, implementing the CRTP protocol. On the firmware side I suggest you ask in the official forum. It's a bit more complicated because the packet is initially received by the NRF51, then transferred over to the STM32 using the syslink.

kashishdhal, 2020-06-03:
Thanks @whoenig , I am closing this issue as I want to analyze all the possible ways suggested. I will touch base if I need more help!

closed 2020-06-03 by kashishdhal