whoenig/crazyflie_ros #183: Problems with crazyflie_ros

eduardo-del, 2020-06-16:
Hi guys i have a problem with crazyflie_ros
when i try tu run hover_vrpn.launch the crazyflie should flie with the next position:
x = 0
y = 0
z = 0.5
using publish_pose.py but the crazyflie can't do the task

 I tried return to install the folder crazyflie_ros but i have the same problem, it can't do the task .
I use the firmware 2020.06 and Optitrack.

I have communication, it fly but crazyflie can't do the task
also the crazyflie work with cflib and other days with crazyflie_ros, but now crazyflie_ros appears don't functionality 

Can some one help me? 
Tank you.

whoenig, 2020-06-17:
Sorry to hear you encountered an issue. Could you be a bit more descriptive on which part works and which one doesn't. When it doesn't work, what is the behavior? Are there any error messages?

eduardo-del, 2020-06-17:
I solved it, the problem was the cameras, I just did a good calibration.
A few minutes ago I finished solving it
Thanks @whoenig

closed 2020-06-17 by whoenig