whoenig/crazyflie_ros #185: Failure to launch crazyflie_server.launch with cf firmware: 2020.06

acxz, 2020-08-30:
Firmware from latest release from github (2020.06) is not working, when trying to launch crazyflie_server.launch.

```
[ INFO] [1598808102.356615935]: Adding radio://0/80/2M as  with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
[ INFO] [1598808102.378432405]: Requesting parameters...
[ INFO] [1598808102.383603020]: Found variables in cache.
[crazyflie_add-2] process has finished cleanly
log file: /home/acxz/.ros/log/a2c0f334-eae3-11ea-adef-0028f8a93e04/crazyflie_add-2*.log
[ INFO] [1598808102.732058682]: Requesting Logging variables...
[ INFO] [1598808102.734889431]: Found variables in cache.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find acc.y in log toc!
[crazyflie_server-1] process has died [pid 59278, exit code -6, cmd /home/acxz/vcs/git/github/mouhyemen/cfm_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/acxz/.ros/log/a2c0f334-eae3-11ea-adef-0028f8a93e04/crazyflie_server-1.log].
log file: /home/acxz/.ros/log/a2c0f334-eae3-11ea-adef-0028f8a93e04/crazyflie_server-1*.log
all processes on machine have died, roslaunch will exit
```

acxz, 2020-08-30:
Seems to be a similar error to https://github.com/whoenig/crazyflie_ros/issues/176

whoenig, 2020-08-30:
Did you manage to fix it either by deleting the cache or making sure that safelink is disabled?

acxz, 2020-08-30:
Yep just had to update the crazyflie_cpp in crazyflie_tools. #186 gets the change into this repo, so that the latest firmware works for people.

PR #186 (2020-08-31): Update crazyflie_tools, disabling safelink
Resolves #185 
add updated crazyflie_tools with updated crazyflie_cpp (disabling safelink)
closed 2020-08-31 by whoenig