whoenig/crazyflie_ros #187: Failing to log valid LOG_GROUP ctrltarget (setpoint)

mouhyemen, 2020-08-31:
I want to log additional log groups from the crazyflie. I modified the scripts to log in **ctrltarget**.

https://github.com/mouhyemen/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp#L207

I get the error message that _ctrltarget.roll_ not found in log toc.

Note that, I tested my modification with another log group **controller** and it works fine. On CF firmware, **ctrltarget** is a valid log group which I also see on cfclient.

mouhyemen, 2020-08-31:
I modified my script where if I log the following, then it works:
`
        logBlockSetpoint.reset(new LogBlock<logSetpoint>(
          &m_cf,{
            {"ctrltarget", "thrust"},
            {"ctrltarget", "yaw"},
            {"ctrltarget", "pitch"},
          }, cb));
        logBlockSetpoint->start(2); // 10ms
`
But if I log `{"ctrltarget", "roll"}` then it says, `what():  Could not find ctrltarget.roll in log toc!`

whoenig, 2020-09-06:
First, you should check if "ctrltarget.roll" also shows up with crazyflie_tools (`listLogVariables`, https://github.com/whoenig/crazyflie_tools/blob/master/src/listLogVariables.cpp). If it does not, it might be a problem with the TOC cache, which you can disable (https://github.com/whoenig/crazyflie_cpp/blob/master/include/crazyflie_cpp/Crazyflie.h#L193). Second, please make sure you have safelink disabled (https://github.com/whoenig/crazyflie_cpp/blob/master/include/crazyflie_cpp/Crazyflie.h#L15).