whoenig/crazyflie_ros #189: Frames

eduardo-del, 2020-09-29:
HI i have a issues with the frames of the Crazyflie and 
the optritracks frame, Do you know how I can make the frames correspond?
I'm using vrpn with optitrack 1.10

I'm trying to do tracking on yaw.

whoenig, 2020-09-30:
Did you try the "Up axis: Z" setting (see https://crazyswarm.readthedocs.io/en/latest/usage.html#optitrack)?

eduardo-del, 2020-09-30:
Hi, thanks for the answer @whoenig  
I'm going to check this [ https://crazyswarm.readthedocs.io/en/latest/usage.html#optitrack](url) in the morning when i'm in the lab.
In this moment i'm writing a code about the topic for roll, pitch and yaw on vrpn_client_ros.cpp

Doesn't matter that if i use crazyflie_ros and not crazyswarm?, this questions is about the link [https://crazyswarm.readthedocs.io/en/latest/usage.html#optitrack](url)

whoenig, 2020-09-30:
The description there applies regardless. Crazyswarm just has OptiTrack support compiled in, while in crazyflie_ros you need to use a separate node/package.

eduardo-del, 2020-09-30:
Hi @whoenig  thanks, all its okay with the frames.

@whoenig  I have another issue, i try to do traking on yaw in publish_pose.py but like do you do know
the angles are in quaternions, do you know how can i give the disered to yaw in publish_pose.py?

I trayed the next code but i had an error

`if __name__ == '__main__':
    rospy.init_node('publish_pose', anonymous=True)
    worldFrame = rospy.get_param("~worldFrame", "/world")
    name = rospy.get_param("~name")
    r = rospy.get_param("~rate")
    o = 0
    x = 0
    y = 0
    z = 0.5
    rate = rospy.Rate(r)

    msg = PoseStamped()
    msg.header.seq = 0
    msg.header.stamp = rospy.Time.now()
    msg.header.frame_id = worldFrame
    msg.pose.position.x = x
    msg.pose.position.y = y
    msg.pose.position.z = z
    quaternion = tf.transformations.quaternion_from_euler(0, 0, 0)
    msg.pose.orientation.x = quaternion[0]
    msg.pose.orientation.y = quaternion[1]
    msg.pose.orientation.z = quaternion[2]
    msg.pose.orientation.w = quaternion[3]
    quaternion = tf.transformations.euler_from_quaternion(0, 0, 0, 0)
    msg.pose.orientation.x = euler[0]
    msg.pose.orientation.y = euler[1]
 
    pub = rospy.Publisher(name, PoseStamped, queue_size=1)

    while not rospy.is_shutdown():	
        msg.header.seq += 1
        o += 0.03
        msg.header.stamp = rospy.Time.now()
	    msg.pose.orientation.z = 90*math.sin(2*3.1416*0.05*o)
        pub.publish(msg)
        rate.sleep()`

in the last month i did a sine way in Z, but in this were easy because just changed the msg.pose.position, but here i have quaternion and i wiil like that the angles will  be euler angles. 

Also the first code that i trayed were it

`    pub = rospy.Publisher(name, PoseStamped, queue_size=1)

    while not rospy.is_shutdown():
        msg.header.seq += 1
        o += 0.03
        msg.header.stamp = rospy.Time.now()
	    yaw = 90*math.sin(2*3.1416*0.1*o)
        pub.publish(msg)
        rate.sleep()`

but the crazyflie didn't the task.

thank you @whoenig

whoenig, 2020-10-01:
there are helper functions for the conversion, see http://wiki.ros.org/tf2/Tutorials/Quaternions#Think_in_RPY_then_convert_to_quaternion.

Glad to hear that the frames work for you now!

closed 2020-10-01 by whoenig