whoenig/crazyflie_ros #190: Issues about hover_vicon.launch

easy98, 2020-10-10:
### #Hello, now I'm confusing about nothing appearing on rviz.

Since I changed its url, when I run the command :"roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/100/250K/E7E7E7E702", it works very well.
But when I run the following:
roslaunch crazyflie_demo hover_vicon.launch
uri:=radio://0/100/250K/E7E7E7E702 frame:=cf2 x:=0 y:=0
z:=0.5
It seemed it didn’t connect to crazyfile. The error is below:

SUMMARY
========
PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: /vicon/crazyflie/...
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/90/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 0.5
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /vicon/datastream_hostport: 192.168.1.100:801
 * /vicon/stream_mode: ClientPull
 * /vicon/tf_ref_frame_id: /world

NODES
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rviz (rviz/rviz)
    vicon (vicon_bridge/vicon_bridge)
  /crazyflie/
    baselink_broadcaster (tf/static_transform_publisher)
    controller (crazyflie_controller/crazyflie_controller)
    crazyflie_add (crazyflie_driver/crazyflie_add)
    joy (joy/joy_node)
    joystick_controller (crazyflie_demo/controller.py)
    pose (crazyflie_demo/publish_pose.py)

auto-starting new master
process[master]: started with pid [11985]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ba7321a6-0a8b-11eb-a0c0-606dc7c60f33
process[rosout-1]: started with pid [11996]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [12003]
process[crazyflie/crazyflie_add-3]: started with pid [12004]
process[crazyflie/joy-4]: started with pid [12010]
process[crazyflie/joystick_controller-5]: started with pid [12016]
[ INFO] [1602288081.535910205]: wait_for_service /add_crazyflie
[ INFO] [1602288081.537635117]: found /add_crazyflie
[ INFO] [1602288081.540543598]: Adding radio://0/90/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 0, Use ROS time: 1
process[crazyflie/controller-6]: started with pid [12022]
process[crazyflie/pose-7]: started with pid [12024]
process[crazyflie/baselink_broadcaster-8]: started with pid [12030]
process[vicon-9]: started with pid [12032]
process[rviz-10]: started with pid [12038]
[ INFO] [1602288081.583060337]: Connecting to Vicon DataStream SDK at 192.168.1.100:801 ...
[ INFO] [1602288081.590780987]: .
[ WARN] [1602288081.720500381]: Couldn't set gain on joystick force feedback: Bad file descriptor
[ INFO] [1602288081.721432046]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ERROR] [1602288082.547704882]: Exception thrown while processing service call: timeout
[ERROR] [1602288082.548137033]: Service call failed: service [/add_crazyflie] responded with an error: timeout
[ INFO] [1602288082.591066603]: ... connected!
[ INFO] [1602288082.591189177]: Setting Stream Mode to ClientPull: Success
[ INFO] [1602288082.591250732]: Axis Mapping: X-Forward Y-Left Z-Up
[ INFO] [1602288082.591327238]: Version: 1.8.0
[ INFO] [1602288082.591365509]: setting up grab_vicon_pose service server ... 
[ INFO] [1602288082.593126335]: setting up segment calibration service server ... 
[ WARN] [1602288082.595192837]: grab frame returned false
[ WARN] [1602288082.641969685]: cf2 occluded, not publishing... 
[ INFO] [1602288082.644535737]: creating new object cf2/cf2 ...
[ WARN] [1602288082.645761735]: unable to load zero pose for cf2/cf2
[ INFO] [1602288082.645797216]: ... done, advertised as " vicon/cf2/cf2" 
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/coven-lab/.ros/log/ba7321a6-0a8b-11eb-a0c0-606dc7c60f33/crazyflie-crazyflie_add-3*.log.

Also, when I run the command: roslaunch crazyswarm hover_swarm.launch
There is nothing in the rviz.

whoenig, 2020-10-10:
* In your log, it stated `Adding radio://0/90/2M as crazyflie` so your argument to change the URI did not succeed. Please double check how you launch. Your command looks correct, but sometimes there can be issues with the spaces etc.
* The crazyswarm requires the datarate set to 2M, not 250K. Also, please make sure you post the output in that case and open an issue on the crazyswarm issue tracker on github.

easy98, 2020-10-10:
Thank you very much!

closed 2020-12-20 by easy98