whoenig/crazyflie_ros #191: Roscpp Update Params

EceChaik, 2020-10-15:
Hi guys,

not really an issue with the package, but I cannot find out how to properly call the UpdateParams service with roscpp, something like the equivalent of:

update_params = rospy.serviceProxy('update_params',UpdateParams)
and
update_params(["kalman/initialX", "kalman/initialY", "kalman/initialZ"])

EceChaik, 2020-10-15:
I made it work by using:

std::vector< std::string > vec;
vec.push_back("kalman/initialX");

and so on, and then

crazyflie_driver::UpdateParams upd_serv;
upd_serv.request.params = vec;
client.call(upd_serv);

it seems to work, but if anyone knows of a better way, do let me know

whoenig, 2020-10-16:
Some other options would be to 1) directly `push_back()` on `upd_serv.request.params`, or 2) use a C++11-style initializer `{"kalman/initialX", "kalman/initialY"}` on `upd_serv.request.params`.

closed 2020-10-16 by whoenig