whoenig/crazyflie_ros #192: Logging Pose Fails

EceChaik, 2020-10-19:
Hi guys,

I have a bunch of crazyflies that I'm starting to mess with, but here's the issue.
In only one of the crazyflies, I have attached the flow deck. 
I am running a simple launch file that just connects to the crazyflie and logs a bunch of info.
However, whenever I enable the "Pose" log as:
<arg name="enable_logging_pose" value="True"/>

the crazyflie with the Flow deck crashes with the message:

terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find stateEstimateZ.quat in log toc!
[crazyflie_server-1] process has died [pid 4059, exit code -6, cmd /home/dc4204/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/dc4204/.ros/log/d6480a7a-11db-11eb-8c75-a0510b0ac9d6/crazyflie_server-1.log].
log file: /home/dc4204/.ros/log/d6480a7a-11db-11eb-8c75-a0510b0ac9d6/crazyflie_server-1*.log

The other crazyflies do not encounter such errors and I'm able to perfectly log their pose.
I'm assuming this has to do with the Flow deck, but I have no idea why including the Flow deck would change the log parameters.
Any ideas?

whoenig, 2020-10-20:
Could you run `rosrun crazyflie_tools listLogVariables` and check if `stateEstimateZ.quat` appears?

EceChaik, 2020-10-20:
Hi, I run that command and it showed some few and far between logs, which seemed very weird.
So I upgraded it with the latest firmware and now it looks like its working. 
Still, have you had any issues with using decks such as the Flow deck along with Vicon?

whoenig, 2020-10-20:
I've never tried using that simultaneously. In principle it should work fine since the flow deck just adds another data source for the EKF. There are some known issues with the flow deck and the uSD card deck though, because both share the same SPI bus.

EceChaik, 2020-10-21:
I guess you can close this then.
By the way, is there like a discussions or other page for such questions?
Cause there are always things that come up that I'm not comfortable opening Github Issues for, since they're more like questions and advice.
Either way, I appreciate the support!

whoenig, 2020-10-23:
It's fine to use "issues" for questions also. If your question is generic, the official support forum is also a good place to get help.

closed 2020-10-23 by whoenig