whoenig/crazyflie_ros #194: Advice on how to use the crazyflie server for low level control

lowjunen, 2020-10-23:
Hey there!

I'm trying to setup a hardware experiment to test some variations of the original Mellinger controller in an Optitrack setup. I've been playing around with the crazyflie_server and crazyswarm_server nodes and I'm currently leaning towards the former as it seems leaner (what without the need to account for multiple drones). I foresee that I'll need some low level access to state estimation too. However, the latter seems to be having more frequent updates. Would you have any advice/tips on which repo to base it from?

Awesome stuff btw!

whoenig, 2020-10-23:
I'd still use the Crazyswarm, because at this point it is better tested and has a larger community. Ultimately, the two projects will merge together. If you need anything that crazyflie_ros has and crazyswarm does not, we can add it for you. Finally, the Crazyswarm tightly integrates Optitrack support in the same node, so the overall latency will be lower (which of course is better for getting the best possible controller performance).

lowjunen, 2020-10-23:
Awesome. Thanks so much!

closed 2020-10-23 by lowjunen