whoenig/crazyflie_ros #196: Cmake error

zhangwiessen, 2020-11-26:
I don't konw how to solve this problem, give me some advice please. thanks

CMake Error at /home/zyl/catkin_ws/devel/share/crazyflie_cpp/cmake/crazyflie_cppConfig.cmake:113 (message):
  Project 'crazyflie_cpp' specifies
  '/home/zyl/catkin_ws/src/crazyflie_ros/crazyflie_cpp/include' as an include
  dir, which is not found.  It does neither exist as an absolute directory
  nor in
  '/home/zyl/catkin_ws/src/crazyflie_ros/crazyflie_cpp//home/zyl/catkin_ws/src/crazyflie_ros/crazyflie_cpp/include'.
  Check the issue tracker 'https://github.com/whoenig/crazyflie_ros/issues'
  and consider creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  crazyflie_ros/crazyflie_driver/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/zyl/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/zyl/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

whoenig, 2020-11-26:
Looks like you are missing the git submodules. Try `git submodule init` and `git submodule update`.

Williamrjw, 2021-07-18:
> Looks like you are missing the git submodules. Try `git submodule init` and `git submodule update`.

I tried that, but a new problem occurred.

```bash
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
USB_LIB
    linked by target "crazyflie_cpp" in directory /root/crazyflie_ws/src/crazyflie_ros/crazyflie_cpp

-- Configuring incomplete, errors occurred!
See also "/root/crazyflie_ws/build/CMakeFiles/CMakeOutput.log".
See also "/root/crazyflie_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
By the way, I am trying to implement this project in docker, and I see that this error seems to be related to USB, so I am considering whether it can be implemented in docker. My code to generate the container

```bash
docker run -it --name testusb --privileged -v /dev/bus/usb:/dev/bus/usb 【IMAGE】 /bin/bash
```

I want to know if this error is related to docker, if it doesn't matter, I sincerely hope you can tell me the solution, if it is related to the characteristics of docker, then I will try other methods, thank you very much!

Williamrjw, 2021-07-18:
> > Looks like you are missing the git submodules. Try `git submodule init` and `git submodule update`.
> 
> I tried that, but a new problem occurred.
> 
> ```shell
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> USB_LIB
>     linked by target "crazyflie_cpp" in directory /root/crazyflie_ws/src/crazyflie_ros/crazyflie_cpp
> 
> -- Configuring incomplete, errors occurred!
> See also "/root/crazyflie_ws/build/CMakeFiles/CMakeOutput.log".
> See also "/root/crazyflie_ws/build/CMakeFiles/CMakeError.log".
> Invoking "cmake" failed
> ```
> 
> By the way, I am trying to implement this project in docker, and I see that this error seems to be related to USB, so I am considering whether it can be implemented in docker. My code to generate the container
> 
> ```shell
> docker run -it --name testusb --privileged -v /dev/bus/usb:/dev/bus/usb 【IMAGE】 /bin/bash
> ```
> 
> I want to know if this error is related to docker, if it doesn't matter, I sincerely hope you can tell me the solution, if it is related to the characteristics of docker, then I will try other methods, thank you very much!

I used
```
apt install libusb-1.0-0-dev
```
catkin_make successfully.
Thank you for your reading this issue

whoenig, 2021-07-20:
Thanks for the update!

closed 2021-07-20 by whoenig