whoenig/crazyflie_ros #197: Reading Ranger Deck Sensor

waterfl0w, 2020-11-26:
Hello, and thank you for your work, :)
I was wondering whether or not there is a topic that is publishing the sensor readings of the ranger deck, or if there is another way to interface with that? I have tried looking around but couldn't find anything like that but it is very well possible that I just missed it since I am new to the world of ROS and C++.

whoenig, 2020-11-27:
Do you mean the z-Ranger or Multi-Ranger deck? In principle, you can always get the data using custom log topics, if you know the logging variable names already. An example launch file is available here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/customLogBlocks.launch.

waterfl0w, 2020-11-27:
I was referring to the MultiRanger deck but if custom log blocks already can do that for me then that is good enough. How would I go about finding out what the variable names are? -  Thank you.

whoenig, 2020-11-27:
To my knowledge there is no good documentation other than the firmware source code (and asking in the forum). It seems you can use https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/range.c#L60-L67, e.g., `range.front`.

waterfl0w, 2020-11-27:
Thank you very much. :+1:

closed 2020-11-27 by waterfl0w