whoenig/crazyflie_ros #202: teleop_xbox360.launch，error occurred

ddwy, 2021-05-09:
During running, the following error occurred：

roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/100/2M/E7E7E7E701

SUMMARY
========

PARAMETERS
 * /crazyflie/crazyflie_add/enable_logging: True
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/100/2M/...
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/quadrotor_teleop/x_axis: 5
 * /crazyflie/quadrotor_teleop/x_velocity_max: 30
 * /crazyflie/quadrotor_teleop/y_axis: 4
 * /crazyflie/quadrotor_teleop/y_velocity_max: -30
 * /crazyflie/quadrotor_teleop/yaw_axis: 1
 * /crazyflie/quadrotor_teleop/yaw_velocity_max: -200
 * /crazyflie/quadrotor_teleop/z_axis: 2
 * /crazyflie/quadrotor_teleop/z_velocity_max: 60000
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /crazyflie/
    crazyflie_add (crazyflie_driver/crazyflie_add)
    crazyflie_demo_controller (crazyflie_demo/controller.py)
    joy (joy/joy_node)
    quadrotor_teleop (crazyflie_demo/quadrotor_teleop)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rqt_plot2 (rqt_plot/rqt_plot)
    rqt_plot3 (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [7584]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f569062a-b06e-11eb-8f61-000c299d8ef1
process[rosout-1]: started with pid [7597]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [7606]
process[crazyflie/crazyflie_add-3]: started with pid [7609]
process[crazyflie/joy-4]: started with pid [7612]
process[crazyflie/quadrotor_teleop-5]: started with pid [7624]
process[crazyflie/crazyflie_demo_controller-6]: started with pid [7666]
[ INFO] [1620527618.407122853]: wait_for_service /add_crazyflie
[ INFO] [1620527618.408546396]: found /add_crazyflie
[ERROR] [1620527618.413334124]: Couldn't open joystick force feedback!
[ INFO] [1620527618.413475019]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
process[rviz-7]: started with pid [7716]
process[rqt_plot2-8]: started with pid [7722]
[ INFO] [1620527618.439297019]: Adding radio://0/100/2M/E7E7E7E701 as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
process[rqt_plot3-9]: started with pid [7732]
[ INFO] [1620527618.552576218]: Requesting parameters...
[ INFO] [1620527618.559591859]: Found variables in cache.
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/ubuntu/.ros/log/f569062a-b06e-11eb-8f61-000c299d8ef1/crazyflie-crazyflie_add-3*.log
[ INFO] [1620527619.247728624]: Requesting Logging variables...
[ INFO] [1620527619.250940105]: Found variables in cache.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find acc.z in log toc!
[crazyflie_server-2] process has died [pid 7606, exit code -6, cmd /home/ubuntu/ros_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/ubuntu/.ros/log/f569062a-b06e-11eb-8f61-000c299d8ef1/crazyflie_server-2.log].
log file: /home/ubuntu/.ros/log/f569062a-b06e-11eb-8f61-000c299d8ef1/crazyflie_server-2*.log
[rqt_plot2-8] process has died [pid 7722, exit code -11, cmd /opt/ros/kinetic/lib/rqt_plot/rqt_plot /crazyflie/battery __name:=rqt_plot2 __log:=/home/ubuntu/.ros/log/f569062a-b06e-11eb-8f61-000c299d8ef1/rqt_plot2-8.log].
log file: /home/ubuntu/.ros/log/f569062a-b06e-11eb-8f61-000c299d8ef1/rqt_plot2-8*.log
[rviz-7] process has finished cleanly

whoenig, 2021-05-12:
It looks like your logging cache might be stale. Please delete your log*.csv files in your `~/.ros` folder and try again.

ddwy, 2021-05-13:
> It looks like your logging cache might be stale. Please delete your log*.csv files in your `~/.ros` folder and try again.

I've tried it once,but the following errors will happen.

`setting /run_id to 7b5af434-b388-11eb-9be4-000c299d8ef1
process[rosout-1]: started with pid [6755]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [6772]
process[crazyflie/crazyflie_add-3]: started with pid [6774]
process[crazyflie/joy-4]: started with pid [6781]
process[crazyflie/quadrotor_teleop-5]: started with pid [6808]
process[crazyflie/crazyflie_demo_controller-6]: started with pid [6814]
process[rviz-7]: started with pid [6815]
[ INFO] [1620868434.054842522]: wait_for_service /add_crazyflie
[ INFO] [1620868434.059550461]: found /add_crazyflie
[ INFO] [1620868434.074709192]: Adding radio://0/100/2M/E7E7E7E701 as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
[ERROR] [1620868434.134494574]: Couldn't open joystick force feedback!
[ INFO] [1620868434.134563865]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1620868434.219783275]: Requesting parameters...
[ INFO] [1620868434.230003728]: Params: 228
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/ubuntu/.ros/log/7b5af434-b388-11eb-9be4-000c299d8ef1/crazyflie-crazyflie_add-3*.log
[ INFO] [1620868435.142632142]: Requesting Logging variables...
[ INFO] [1620868435.146542661]: Log: 465
[INFO] [1620868435.261383]: found update_params service
[INFO] [1620868435.261684]: waiting for emergency service
[INFO] [1620868435.263056]: found emergency service
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find acc.x in log toc!
[crazyflie_server-2] process has died [pid 6772, exit code -6, cmd /home/ubuntu/ros_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/ubuntu/.ros/log/7b5af434-b388-11eb-9be4-000c299d8ef1/crazyflie_server-2.log].
log file: /home/ubuntu/.ros/log/7b5af434-b388-11eb-9be4-000c299d8ef1/crazyflie_server-2*.log
`

whoenig, 2021-05-15:
Can you run `rosrun crazyflie_tool listLogVariables --uri radio://0/100/2M/E7E7E7E701` and check if `acc.x` is listed there?

Shirleytyf, 2021-05-16:
when I run **rosrun crazyflie_tools listLogVariables --uri radio://0/80/2M/0xE7E7E7E7** , it shows timeout, when I run **rosrun crazyflie_tools listLogVariables** acc.x is listed.

Shirleytyf, 2021-05-16:
I renumbered it.When I run **rosrun crazyflie_tools listLogVariables --uri radio://0/80/2M/0xE7E7E7E7E1** acc.x and acc.y are listed. 
During running, the following error occurred：
roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/80/2M/E7E7E7E7E1
SUMMARY
========

PARAMETERS
 * /crazyflie/crazyflie_add/enable_logging: True
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M/E...
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/quadrotor_teleop/x_axis: 5
 * /crazyflie/quadrotor_teleop/x_velocity_max: 30
 * /crazyflie/quadrotor_teleop/y_axis: 4
 * /crazyflie/quadrotor_teleop/y_velocity_max: -30
 * /crazyflie/quadrotor_teleop/yaw_axis: 1
 * /crazyflie/quadrotor_teleop/yaw_velocity_max: -200
 * /crazyflie/quadrotor_teleop/z_axis: 2
 * /crazyflie/quadrotor_teleop/z_velocity_max: 60000
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rqt_plot2 (rqt_plot/rqt_plot)
    rqt_plot3 (rqt_plot/rqt_plot)
    rviz (rviz/rviz)
  /crazyflie/
    crazyflie_add (crazyflie_driver/crazyflie_add)
    crazyflie_demo_controller (crazyflie_demo/controller.py)
    joy (joy/joy_node)
    quadrotor_teleop (crazyflie_demo/quadrotor_teleop)

auto-starting new master
process[master]: started with pid [20888]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 53425d96-b619-11eb-b269-3cf862e573dd
process[rosout-1]: started with pid [20899]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [20906]
process[crazyflie/crazyflie_add-3]: started with pid [20907]
ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/ypp/crazyflie_ws/src
ROS path [2]=/opt/ros/melodic/share
process[crazyflie/quadrotor_teleop-5]: started with pid [20913]
process[crazyflie/crazyflie_demo_controller-6]: started with pid [20919]
process[rviz-7]: started with pid [20925]
process[rqt_plot2-8]: started with pid [20926]
[ INFO] [1621150546.170649352]: wait_for_service /add_crazyflie
[ INFO] [1621150546.174141873]: found /add_crazyflie
[ INFO] [1621150546.179627610]: Adding radio://0/80/2M/E7E7E7E7E1 as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
process[rqt_plot3-9]: started with pid [20927]
[ WARN] [1621150546.309762982]: Received unrequested data for block: 87
[ WARN] [1621150546.311247078]: Received unrequested data for block: 87
[ INFO] [1621150546.312106069]: Requesting parameters...
[ INFO] [1621150546.317560038]: Found variables in cache.
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/ypp/.ros/log/53425d96-b619-11eb-b269-3cf862e573dd/crazyflie-crazyflie_add-3*.log
[ INFO] [1621150546.712558824]: Requesting Logging variables...
[ INFO] [1621150546.715569613]: Found variables in cache.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find acc.y in log toc!
[INFO] [1621150546.764116]: found update_params service
[INFO] [1621150546.765897]: waiting for emergency service
[INFO] [1621150546.769599]: found emergency service
[crazyflie_server-2] process has died [pid 20906, exit code -6, cmd /home/ypp/crazyflie_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/ypp/.ros/log/53425d96-b619-11eb-b269-3cf862e573dd/crazyflie_server-2.log].
log file: /home/ypp/.ros/log/53425d96-b619-11eb-b269-3cf862e573dd/crazyflie_server-2*.log